intel-aero/intel-aero-docs #2: Create a ROS meaningful sample using docker container

anselmolsm, 2017-08-02:
The idea is to provide a sample that shows the use of containers and also uses ROS features, more than just a ros tool execution.

shakthi-prashanth-m, 2017-08-10:
@anselmolsm : basically the usage of docker for ROS is
```
root@intel-aero# docker pull ros
root@intel-aero# docker run -it ros
```
which would pull ROS & run roscore on Aero platform; this leads to openup ros interactive shell.
Developers can run `roscore` in the ROS interactive shell and also run apps.
Generally, ROS apps are developed @ host machine for.eg: Ubuntu. Ros has tutorial for developing applications [here](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29). It is easy for them to follow it.
Please guide what exactly to be documented as part of Intel Aero Wiki.

Thanks

shakthi-prashanth-m, 2017-08-10:
Its good idea to demonstrate a sample application (using MAVROS) that talks to FC. Isn't it ?

anselmolsm, 2017-08-11:
@shakthi-prashanth-m I think that's a way to go, indeed.

shakthi-prashanth-m, 2017-08-16:
Hi Anselmo, I've created MAVROS App for controlling Aero FC. Kindly review: https://github.com/intel-aero/ros-examples/pull/1
Thanks