intel-aero/intel-aero-docs #25: Add FPGA/FC recovery instructions

lucasdemarchi, 2017-08-09:
Talk about the different FPGA firmwares, particularly the options for recovery (rtf-recovery.jam and aerofc-force-bootloader.py)

sugnanprabhu, 2017-08-11:
@lucasdemarchi details about the jam files are already documented under the following section. 
https://intel-aero.github.io/getting-started/update/#fpga
Is it ok to add details of aerfc-force-bootloder.py under this section?

lucasdemarchi, 2017-08-11:
No. Here we need to fully detail the recovery mechanisms, going from the simpler/faster one (force bootloader pin) to the more complex one (fpga recovery firmware).

lucasdemarchi, 2017-08-11:
We can add a link to this instructions from there, but this is certainly not part of "getting started"

sugnanprabhu, 2017-08-14:
created PR #33 
@lucasdemarchi I have added the instruction under troubleshooting, please let me know if it has to be moved under different section.

anselmolsm, 2017-08-16:
Does https://github.com/intel-aero/meta-intel-aero/wiki/91-%28References%29-FPGA belong here? Or should we create a separate section for that?

lucasdemarchi, 2017-08-16:
> Does https://github.com/intel-aero/meta-intel-aero/wiki/91-%28References%29-FPGA belong here? Or should we create a separate section for that?

I think that should be in a "development / fpga" section, not here.

lucasdemarchi, 2017-08-16:
Btw, in that wiki page the "you need to format as vfat"  is wrong and should be removed.