intel-aero/intel-aero-docs #29: mkdocs: add easy way to add captions to images

lucasdemarchi, 2017-08-10:
See "TODO" in #23.  One option that may be good: https://github.com/jdittrich/figureAltCaption

sugnanprabhu, 2017-08-14:
@lucasdemarchi 
The above mentioned extension did not work with the mkdocs.
I have included the following markdown instead and submitted a PR #32 .
https://github.com/helderco/markdown-figures

If the changes look ok, will make the required changes in all the current pages.

```
A simple example:
  ![comment](img/figure.png)
  : figure caption

Outputs:

  <figure>
    <img alt="" src="img/figure.png" />
    <figcaption>
      <p>figure caption</p>
    </figcaption>
  </figure>
```

sugnanprabhu, 2017-08-14:
Following is the resulting format achieved after using this markdown

```
![qgc-calibration02](img/calibration_02.png)
: Reset to defaults

![qgc-calibration03](img/calibration_03.png)
```
[image]