intel-aero/intel-aero-docs #38: Development/Flight Stacks

anselmolsm, 2017-08-15:
https://intel-aero.github.io/dev/flight-stacks/

How to flash the recommended/tests firmware shipped in the image:
`aerofc-update.sh /etc/aerofc/px4/<FS>.px4`

A guide for developers who want to develop the flight stacks:
 - link to the upstream projects
 - instructions on how to build and flash a custom firmware version:
    - PX4: make aerofc-v1 upload
    - ArduPilot: ./waf configure --board aerofc-v1 && ./waf --target bin/arducopter --upload

avinash-palleti, 2017-08-16:
@lucasdemarchi @anselmolsm Please review the commit once,
https://github.com/avinash-palleti/intel-aero-docs/commit/864b8a38c72b7e5f7222ff41cacfdbddf79f2d8b

Let me know if anything needs to be added, otherwise i will submit PR.

lucasdemarchi, 2017-08-16:
Can you send it as PR?

avinash-palleti, 2017-08-16:
https://github.com/intel-aero/intel-aero-docs/pull/44

PR #44 (2017-08-16): Add Flightstack development document
Closes #38
closed 2017-08-16 by anitha-suresh